utra-robosoccer/soccer-embedded #61: Development Plan for new Communication Interface and C++ Refactoring

rfairley, 2018-08-26:
# Development Plan for new Communication Interface and C++ Refactoring

This issue is to discuss and determine how to go about implementing two major changes to `Robot`:
- developing the new [Communication Interface](#36)
- reorganizing existing code to take advantage of C++ features

## Goals

These are some goals the development plan could have and why they would benefit us:

- team coordination and work in parallel where possible
    - speeds up development process
    - ensures everyone knows a bit about the system
- maintain a stable functional (works on hardware) version of `Robot` we build upon incrementally
    - provides a sanity check - can always revert to something that works
    - avoids rewriting - work with existing code
- build a foundation of unit tests/mocking
    - carries forward and checks for correctness in future code growth

## Plan (proposed, pending discussion)

1. Finalize UML class diagram of Communication Interface
    - discuss this next week
2. For each `FreeRTOS Task`, list how the refactored code accomplishes the same functionality
    - this will confirm that we gave all classes necessary responsibility and accounted for all functionality
3. **discussion needed** There are several ways we could go about refactoring.

    Initially I had in mind starting by matching the existing functionality of receiving a PC goal, sending commands to motors and reading from sensors, and sending back to PC. However, to implement this would require almost the entire code path of the Communication Interface be completed (including `EquipmentQueue`, `SensorDataCache`, ...) - which does not tie in with incrementally adding in new features.

    An alternative development path is to preserve some minor functionality, add the essential classes in that follow the new code path, and verify it works on the hardware. We make sure we unit test and mock those classes before moving on to add more functionality.

    For example, let's say we preserve the following functionality: sending the motor commands `cmdReadPosition` to motors on UART, waiting for the reading, and transmitting that to the PC. We can adapt the code that does this to match the system in https://github.com/utra-robosoccer/soccer-embedded/issues/36#issuecomment-410638616 by adding in a timer, `Motor read cmd generator`, `Cache writer`, `Sensor data cache` (with position data only), `EventHandler` and `PCInterface` (the Transmit parts only). To test on hardware, we have the new motor readings from the cache sent on a timed basis (in the final system this is event based). We then work our way across the rest of the diagram in https://github.com/utra-robosoccer/soccer-embedded/issues/36#issuecomment-410638616, adding unit tests as we go and verifying on the hardware.

    The example above may not be the best way to start - another alternative is start from the Receive and RX event handling functionality and return placeholder sensor data from the cache. There are many places on the diagram https://github.com/utra-robosoccer/soccer-embedded/issues/36#issuecomment-410638616 we can start coding at.

tygamvrelis, 2018-08-26:
> An alternative development path is to preserve some minor functionality, add the essential classes in that follow the new code path, and verify it works on the hardware. We make sure we unit test and mock those classes before moving on to add more functionality.

Yes.

One other thing I just want to state here for the record (which, if I recall correctly, was brought up at the meeting) is that we can refactor our current code base a bit more to "prepare" it for these changes. For example, we have a nasty 100-line routine for our RX thread right now. I am guessing this could probably be broken down into <10 lines if we create clever functions, which would be implemented in a single file which contains all the RX application code.

This is more similar to how it might look when it is object-oriented (since the functions we would have to create would form an interface of sorts), and it will help us eliminate global variables and other weird dependencies that might current exist.

rfairley, 2018-09-04:
Yes - the preparation step had slipped my mind from our discussion. 

> I am guessing this could probably be broken down into <10 lines if we create clever functions

I like the sound of this - refactoring into functions first. We should be able to reuse those functions, and bring them into the new system as needed as we build up the new code paths. That way, the reusable parts of the "application-level" functionality are preserved.

rfairley, 2018-09-04:
I imagine our roadmap will look like this - thoughts?

[image]

Attached XML to edit (remove .txt from the extension):

[comm_interface_dev_path.xml.txt](https://github.com/utra-robosoccer/soccer-embedded/files/2347007/comm_interface_dev_path.xml.txt)

tygamvrelis, 2019-03-03:
@rfairley Can we close this?

rfairley, 2019-03-03:
Yes - won't be changing the diagram anymore, so is good to close

closed 2019-03-03 by rfairley