utra-robosoccer/soccer-embedded #71: Need to create mockable classes for hardware functions

tygamvrelis, 2018-09-13:
**Describe the feature**
Each hardware module we use needs to have a mockable class. These classes will basically just consist of trivial virtual methods which wrap function calls to ST's libraries. Since these functions are virtual, they can be mocked with gmock.

**Reason for request**
Allows us to create tests for our code without requiring the microcontroller hardware to execute them.

**Timeline**
Sooner is better.

**Peripherals to be mocked**
Add to this if you see anything that was missed.
- [ ] Ethernet
- [ ] UART
- [ ] I2C

tygamvrelis, 2018-09-14:
@rfairley is working on Ethernet and I am working on UART. Rest is up for grabs

rfairley, 2018-09-15:
One thing I completely forgot about when making the virtual UdpInterface class is that virtual classes do require accessing vtables to locate the correct sub class member function. The cost of the accesses probably isn't too high if the vtables are stored in static memory, but we may as well avoid the extra accesses. We can get around this using templates, https://github.com/google/googletest/blob/master/googlemock/docs/CookBook.md#mocking-nonvirtual-methods.

rfairley, 2018-09-16:
Right now this branch has the templating added https://github.com/utra-robosoccer/soccer-embedded/tree/rfairley-pcinterface. One issue we run into is the member function definitions for a templated class need to be in the same translation unit - usually this is the `.h` file for that class. There are a couple of other workarounds, like including the .cpp file in the .h or explicitly instantiating the class with a specific template type somewhere in the code (https://code.i-harness.com/en/q/858ec5). Those seem a bit hacky though - `.h` is probably the safest way.

rfairley, 2018-09-29:
Update on Ethernet mockable class: now in progress on rewriting `UdpInterface`, its subclasses, and `PcInterface` which uses its subclasses with functions that take parameters (currently the API wrapper functions take no functions and pass in data members). Having the functions take parameters will allow gmock checking of parameters in testing.

Had problems with getting Lwip dependencies in earlier, so that types like `struct pbuf*` are defined:
- Copying over the Lwip dependencies from another H7 project ends up needing the rest of HAL. Tried this, but was unsuccessful to get the build to work (probably some setting somewhere Cube sets but I missed).
- Including the deps from `Robot` - this project is for the F4 board, and Lwip is not included as a middleware to add - so this cannot be done for the `LwipUdpInterface` class.

The following solved the problem:
- Include the deps from `Development/lwip-rtos-config` by adding an include path in the eclipse project settings. This builds (with `LwipUdpInterface.cpp` excluded from the build). This requires having `lwip-rtos-config` loaded in the workspace.

closed 2018-11-10 by tygamvrelis