utra-robosoccer/soccer-embedded #78: Testing framework integration is incomplete

tygamvrelis, 2018-09-16:
# Motivation
We would like to be able to switch between running our unit tests and compiling code for the microcontroller with as little effort as possible. Furthermore, we have found that having a separate project just for testing is quite difficult to set up and maintain, since this requires sharing source files and header between projects which is a bit messy.

# Action items
We identified during today's meeting that it should be possible to use just 1 project for both testing and development through the magic of _run configurations_. A run configuration basically specifies which compiler you use, what include paths and libraries you use/link to, and which preprocessor macros are defined.

**Unit testing configuration**
- Will use MinGW compiler
- Will not compile any HAL or FreeRTOS libraries (it can't compile these, really)
- Will use the GTest unit test runner to display test results
- Will define a macro which includes the Google test/mock framework main and excludes the microcontroller program main
- [ ] Unit testing configuration created & tested

**Microcontroller configuration**
- Will use the STM32 Toolchain
- Will not compile Google test framework libraries
- Will define a macros which excludes the Google test/mock framework and includes the microcontroller program main
- [ ] Microcontroller configuration created & tested

rfairley, 2018-10-01:
Was experimenting with the build configurations, while adding the UdpInterface-related files to Development/F7_Validation/Robot. Adding a new configuration called "`Test`", and doing the following seems to work:

- Have `Test` use MinGW GCC and CDT under `C/C++ Build` > `Toolchain Editor`. Got internal errors after changing this, but but after restarting Eclipse it was OK.
- Under `C/C++ General` > `Preprocessor Include Paths, Macros, etc` > `Providers`, uncheck `Ac6 SW4 ... Built-in compiler settings`. Seems this stops the builder from wanting to use the special MCU instructions.
- Exclude any `*_test.cpp` files from `Debug` build
- Exclude hardware resources from `Test` build - this includes `Drivers`, `Middleware`, lots of files in `Inc` and `Src`, and `startup`
- Make sure the setup from https://github.com/utra-robosoccer/soccer-embedded/wiki/Setting-Up-Google-Test-(gtest)-and-Google-Mock-(gmock) is complete

This is what it looks like after building:
[image]

The `Robot.elf` gets created, but the overall build fails because `arm-none-eabi-objcopy` is not found. But the `.elf` is all that is needed to run the tests (guessing `arm-none-eabi-objcopy` is something to do with converting the elf to the target ARM arch). Setting up a run configuration for C/C++ Unit with the path to the `Test/Robot.elf` file runs the tests successfully.

Would be nice to have `Test` report the build as successful. To do this we would need to find a way for the toolchain that the `Test` build config uses to stop calling `arm-none-eabi-objcopy` in the build process.

gokuldharan, 2018-10-06:
Due to the restructuring, a separate project using linked resources to source files under Common/ seems to be the best approach. This eliminates the need for working around build configurations and also allows us to test with board-specific drivers by using a configuration for each board. This means we can include everything in Drivers/, which makes the testing more applicable to the use-case. It is also more extendable than the approach where testing is built into the project, as every new source file will have to be excluded from the test configuration whereas in RobotTest we exclude everything in Common by default, and you only have to uncheck the exclude option once they are compatible with our testing framework. Finally, this also fits better with the new code structure in my opinion, with all high-level source files under Common/ and any clients of these files linking to them with linked resources. It also allows us to have all of our test files in one place, namely under Common/test/.

closed 2018-10-07 by tygamvrelis