utra-robosoccer/soccer-embedded #90: language_settings.xml is under version control, which adds clutter to code reviews

tygamvrelis, 2018-09-26:
Please see @rfairley's comment in #85 regarding the file `language_settings.xml`, repeated below for convenience:

> Seems the file is still getting tracked by git though. I think git rm --cached Robot/.settings/language.settings.xml is needed to untrack the file - then the rule we have in .gitignore should be good.

Let us try to untrack the file, starting with @rfairley's proposed course of action

rfairley, 2018-11-03:
we might consider adding the `.settings/org.eclipse.cdt.{core,ui}.prefs` files to version control, to track IDE style settings (or explicitly not track these by adding these files to the gitignore). maybe decide on this once code is settled

rfairley, 2019-01-20:
Closing with #168

closed 2019-01-20 by rfairley