utra-robosoccer/soccer-embedded #121: Maintenance Tracker

tygamvrelis, 2018-11-09:
# Purpose
This issue is where we can track all our ideas for what we want to improve after Dec. 1.

# Items
- [ ] Change all FreeRTOS calls to the CMSIS equivalents
    - **Reason**: It is more portable as CMSIS is a standard API. They also have some pretty useful APIs
- [ ] Define peripheral handles in some unified location (e.g. instead of using `&huart5` in calls to PC interface functions, we would instead use something like `PC_USART_HANDLE`
    - **Reason**: This will it easier to deploy our code onto the different boards we support (F4, F7)
- [ ] Decide on file name and namespace conventions
    - **Reason**: Better to have this style be more consistent
- [ ] Decide on a few key namespaces to group things by
    - **Reason**: As we grow our code base, we will have new peripherals, data structures, algorithms, etc. We should agree on namespaces these things should go in, otherwise things will get hard to maintain fast
- [ ] Consider making `THREADED` a compiler option (-D) as opposed to a macro defined in a common header
    - **Reason**: Some tests may inadvertently include the system header through some dependency chain, but they require a different value for `THREADED` (e.g. they require it to be defined but it is not, or vice versa)
- [ ] Integrate a continuous integration (CI) service for automated builds and unit testing (for pull requests)
    - **Reason**: Better development flow
- [ ] Set up some sort of server that we can log into if we want to test our code at runtime
    - **Reason**: This is long overdue. It's very important that it's easy for people to test their code at runtime as this is always the final check and it is inconvenient for everyone to have to do it in person (additionally there can be a lot of competition if everyone tries to do that during our weekly meetings)
- [ ] Add a CI linter (code style checker)
    - **Reason**: Improve code consistency, and can have checks that make sure `malloc` and `new` are not used (can be missed in code reviews)
- [ ] Consider separating threads into different files
    - **Reason**: Hopefully less merge hell if high-level application code is managed in separate files as opposed to just `freertos.cpp`
- [ ] Remove names after `@file` in Doxygen
    - **Reason**: Apparently we do not actually need any name after this tag, so we should just remove it using a script
- [ ] Consider separate TX and RX threads for PC program
    - **Reason**: It just seems natural to consider these as separate execution contexts, and maybe it could open up the door for some cool opportunities. Should explore it a bit

tygamvrelis, 2018-11-09:
- [ ] Consider using a watchdog timer (WDT) to recover from critical faults
    - **Reason**: if we have an unrecoverable system fault, we need to be able to recover
- [ ] Investigate low-power modes (and how to do this with FreeRTOS)
    - **Reason**: we know our system is busy, but it _does_ (as of this writing) spent time idle. We should see if we can enter a low-power state to save power consumption
- [ ] Create permanent logs
    - **Reason**: it would be great if we could log events (particularly _important_ ones) in a non-volatile memory. We may need support from the electrical team to get an EEPROM on the PCB (probably would use SPI)

rfairley, 2018-11-11:
- [ ] check convention of <> for library file includes and "" for project file includes is applied consistently throughout project

rfairley, 2018-11-12:
- [ ] way to ensure CubeMX settings are synced across boards
  - e.g. common components such as FreeRTOS should have equivalent settings whether on F4 or F7. if Cube settings change on one board, they should also change on the other

tygamvrelis, 2018-11-12:
- [ ] Make sure all private members are using the prefix `m_`
    - **Reason**: Conforms better to our style guide

tygamvrelis, 2018-11-15:
- [ ] Allow different IO type to be used for TX and RX in UartDriver
    - **Reason**: No reason why not. The IO type does not need to be the same for both, it really depends on the requirements of the application

rfairley, 2018-11-16:
- [ ] Check if `volatile` needs to be used in some places throughout application code (mentioned by @tygamvrelis )
  - **Reason**: if we do a release build with optimizations, compiling with `O3` might subtly change the critical parts of the code. We should verify that `volatile` has been used with this in consideration.

rfairley, 2018-11-17:
- [ ] make sure type sizes are explicitly declared (e.g. `uint32_t` instead of `int`). Applies to enumerations and any other higher-level types that have an underlying numeric type as well.
    - **Reason**: best practice when compiling embedded code. Mostly we have been good about doing this, but this may have been missed for the less-obvious types.

rfairley, 2018-11-17:
- [ ] check for unnecessary #includes and remove
    - **Reason**: taking things out of source files and refactoring may have left some includes that are no longer used. They shouldn't cause a problem but it is unneeded preprocessor logic.

rfairley, 2018-11-24:
- [ ] Split out testable components such as `UartInterface`, `OsInterface`, `UdpInterface` from the project-specific `Common/includes` and `Common/hardware` folders. This involves moving the source files somewhere that they can be easily included by projects that are not necessarily related to `Robot`. It might look like a repository imported as a submodule like `utra-robosoccer/soccer-testable-hardware` or `utra-robosoccer/soccer-hardware-mocks`.
  - **Reason**: Projects like omnibot may benefit from this if we want to quickly add unit tests. We should be able to move around project-agnostic files easily when starting a new project, e.g. only one folder of `includes` needs importing to get the full suite of testable interfaces.
  - This goes hand in hand with rethinking the namespaces. Components should be imported individually, e.g. like how gtest has the client import `::testing::Return`. Though, this work to make the name spaces fit for "public" projects need not be done as part of the item `Decide on a few key namespaces to group things by`. This task can happen further down the road.

rfairley, 2018-11-25:
- [ ] A standard way to patch the generated files by CubeMX. Might take the form of a `.patch` file that we automatically apply somehow while building. We can find out how much files have deviated from generated files already by running `git diff --patch -- <path to vanilla generated file> <path to edited generated file>`.
  - **Reason**: It is still beneficial to quickly go through the flow of 1) change a parameter or add something in CubeMX, 2) regenerate the file, and 3) apply edits we may have manually made to the file. Comes up in #141 when `usart.c` needs to be modified with custom edits.

tygamvrelis, 2018-12-15:
- [ ] Style purge—revise the programming style guide to make it much shorter. Make the focus on writing code that has a consistent look and feel, then purge all our code to follow these look and feel rules (maybe via an automated tool...?)

rfairley, 2019-01-14:
Ensure doxygen comments are made at the function declarations in header files, not source file definitions.
 - Reason: should be consistent; there are some files where I missed this.

rfairley, 2019-02-10:
-  [ ] Automatic indentation for eclipse project
  - Reason: more easily format code to our chosen indentation. Eclipse already has a builtin Allman style formatter, which we can adapt if needed

rfairley, 2019-02-10:
- [ ] Update include paths in .cproject file after unified Robot changes (#171)
  - Reason: not needed for building as we are now using the makefile, but we should make sure these are synced so the IDE can index the code properly

rfairley, 2019-02-16:
- [ ] (noting from #169): Try out using doxygen TODOs in comments, and decide if we want to mandate using the Doxygen style TODO tag in addtion to the traditional TODO:
  - Can start from https://stackoverflow.com/questions/8535635/can-doxygen-easily-be-configured-to-recognise-todo-and-fixme-lines and http://www.doxygen.nl/manual/commands.html#cmdtodo
  - Need to update the CONTRIBUTING.md doc if we decide to use `\todo`.

rfairley, 2019-02-16:
- [ ] Think about where test fixtures can be better utilized in unit tests.
  - This applies to `UdpDriver_test.cpp` and `UartDriver_test.cpp` really. The tests are correct in declaring new mock instances for each test and a new `Udp/UartDriver` per test. We can probably optimize and have fewer declarations overall though, by declaring common base `Udp/UartDriver` options in a fixture.