utra-robosoccer/soccer-embedded #122: Tutorial Tracker

tygamvrelis, 2018-11-09:
# Purpose
We can use this issue to track tutorials that we want to make

# Items
- [ ] C++ project setup in System Workbench
- [ ] STM32: Getting started with Ethernet
- [ ] Getting started with PySerial (maybe multithreaded version too?)
- [ ] STM32: FreeRTOS Diagnostics ([trace](https://www.freertos.org/rtos-trace-macros.html) and [run time stats](https://www.freertos.org/rtos-run-time-stats.html))
- [ ] STM32: Debugging with STMStudio

Whenever we figure out how to make our Cube + System Workbench flow work with CI tools (or whenever we figure out how to _adapt_ this flow to work with CI tools), we should make a tutorial on that as well.

rfairley, 2018-11-10:
another how-to idea for reference:

[ ] regenerating code from CubeMX - which parts to manually sync to Robot (e.g. deleting `freertos.c`, copying semaphore definitions)

rfairley, 2018-11-11:
wiki page for future:

- [ ] deviations from the original design in our implementation + reasons (basically a Lessons Learned)