utra-robosoccer/soccer-embedded #150: "pure virtual method called" when EXPECT_CALL of a function that doesn't get called

rfairley, 2018-12-07:
**Describe the bug**
When an `EXPECT_CALL` is given with `.Times(1)`, sometimes the test results show as green and the console has the following printed:

```
pure virtual method called
terminate called without an active exception
```

An example of this occurring is the message shows in one of the UDP driver tests (`UdpDriverShould, FailReceiveRxPbufNull`) when the following line is given (will be visible in master branch soon):

```
EXPECT_CALL(udp_if, pbufCopyPartial(_, _, _, _)).Times(1).WillOnce(Return((u16_t) 1));
```

**To Reproduce**

Insert the following line at the start of `TEST(UdpDriverShould, FailReceiveRxPbufNull){`

```
EXPECT_CALL(udp_if, pbufCopyPartial(_, _, _, _)).Times(1).WillOnce(Return((u16_t) 1));
```

then build and run `Test_F7` in `RobotTest`.

This test is not currently in master, will update once it is.

**Expected behavior**
The test run should have a "red" status and show a red bar and indicate that the method given in `EXPECT_CALL` wasn't called enough times. Right now, the `pure virtual method called` message in the console is the only indicator of something not right.

**Additional context**
Have hit this a few times in the `UdpDriver` tests only and am not sure if it is a problem of ours or of gtest yet. It doesn't seem to happen consistently either (i.e., the red bar and the correct report of "test expected to be called 1 time but called 0 times" does show). Posting this issue so it won't be forgotten, but will update with better details to reproduce once the updated UDP unit tests are in master.

An example where the same situation did give correct behavior is adding the same line to the beginning of `TEST(UdpDriverShould, FailReceiveRxArrayNULL)`.

tygamvrelis, 2018-12-07:
I just looked this issue up and people are saying it is a generic C++ issue which can happen when you call a virtual method from a constructor or destructor. That would seem to be a different issue than what you described (it is generally dangerous to call virtual functions in the constructor). Can you please verify that this is not the issue?

rfairley, 2018-12-08:
Thanks for looking it up, yeah - no virtual calls are being made in the destructor or constructor of the base class (`UdpInterface`, which `LwipUdpInterface` and `MockUdpInterface` inherit from).

Seems like since the tests are being reported as passing, then gtest got at least that far and the problematic virtual method call is being made somewhere in the tear-down process.

One other instance of strange behavior I saw yesterday was one test failing (for a legitimate reason) as logged by gtest but reporting the test **after** it which should have passed as failed (i.e. marking the wrong test as failed). After moving the global declarations of mock interfaces that I was using across all tests (specifically `MockUdpInterface` and `MockOsInterface`) to locally scoped declarations, declared within the `TEST` block, this issue went away. I'm guessing since gtest tracks results associated with the function calls made on the `Mock...` object after using `EXPECT_CALL`, it was getting mixed up between tests due to using the same instance across multiple tests.

Will go back and get the concrete code examples so it can be reproduced.

rfairley, 2018-12-08:
But yeah, probably just user error here, probably should be using a fixture for the `Mock...` interfaces rather than a global instance.

rfairley, 2019-02-07:
Did some investigation to get reproducible examples of erroneous behavior when using global instances of the Mock* interfaces.

Doing something like the following diff (see the Details tab), tests start failing for incorrect reasons.

E.g., the UdpDriver_test.cpp test FailTransmitWhenTxBuffNull fails with pbufFree() called 4 times (expected 1 time). This function can only possibly be called 1 or 0 times from looking at the code.

My guess is that gtest keeps some state (e.g. which functions were called how many times, did the test pass or fail, ...) within the Mock objects. When a single global Mock interface is used, the state gets carried over from one test to another instead of being reset before/after each test. If using a new object for each test, the state is correctly reset by destroying and recreating a new mock object.

This behavior can be seen by trying the branch at https://github.com/utra-robosoccer/soccer-embedded/commit/f0b66cb9d8ae27d358c69a2bcff9eaae1d0ff219.

To see the `pure virtual method called` error message, the branch at https://github.com/utra-robosoccer/soccer-embedded/commit/f8dfd528089231f1132d85872ba36b7f48602695 can be tried [edited at line 224](https://github.com/utra-robosoccer/soccer-embedded/commit/f8dfd528089231f1132d85872ba36b7f48602695#diff-362979770b00d124384dc0c3cde908fbR224) in attempt to make only this test fail. To see gtest correctly handle that failure (using one Mock object instance per test), see this branch: https://github.com/utra-robosoccer/soccer-embedded/commit/4cddf7eddd6d2e7758afbc2abe81808d8339620a.

With this in consideration, we should not use global Mock instances for unit tests, and either 1) construct a new Mock instance per test, or 2) have the Mock instance exist in a test fixture so that gtest/gmock can correctly reset its state `CircularDmaBuffer_test.cpp` [does this right now](https://github.com/utra-robosoccer/soccer-embedded/blob/b75bb231856372494d94c2a5009d3b9591efbf10/Common/test/CircularDmaBuffer_test.cpp#L59).

<details>
```diff
diff --git a/Common/test/UdpDriver_test.cpp b/Common/test/UdpDriver_test.cpp
index 8d6a71b9..25806f3d 100644
--- a/Common/test/UdpDriver_test.cpp
+++ b/Common/test/UdpDriver_test.cpp
@@ -50,12 +50,12 @@ namespace {
 // Constants
 // ----------------------------------------------------------------------------
 const ip_addr_t ZERO_IP_ADDR_T = {0x0};
+MockUdpInterface udp_if;
+MockOsInterface os_if;
 
 }
 
 
-
-
 TEST(UdpDriverShould, DefaultInitializeMembersToZero) {
     UdpDriver udpDriverUnderTest;
 
@@ -68,8 +68,7 @@ TEST(UdpDriverShould, DefaultInitializeMembersToZero) {
 }
 
 TEST(UdpDriverShould, InitializeMembersWithParameterizedConstructor) {
-    MockUdpInterface udp_if;
-    MockOsInterface os_if;
+
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
     const ip_addr_t TEST_IP_ADDR_PC = {0xC0A80002};
 
@@ -85,8 +84,8 @@ TEST(UdpDriverShould, InitializeMembersWithParameterizedConstructor) {
 }
 
 TEST(UdpDriverShould, SucceedInitialize) {
-    MockUdpInterface udp_if;
-    MockOsInterface os_if;
+
+
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
     const ip_addr_t TEST_IP_ADDR_PC = {0xC0A80002};
 
@@ -100,7 +99,7 @@ TEST(UdpDriverShould, SucceedInitialize) {
 }
 
 TEST(UdpDriverShould, FailInitializeWhenUdpInterfaceNull) {
-    MockOsInterface os_if;
+
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
     const ip_addr_t TEST_IP_ADDR_PC = {0xC0A80002};
 
@@ -111,7 +110,7 @@ TEST(UdpDriverShould, FailInitializeWhenUdpInterfaceNull) {
 }
 
 TEST(UdpDriverShould, FailInitializeWhenOsInterfaceNull) {
-    MockUdpInterface udp_if;
[... 146 lines cut ...]
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
@@ -339,8 +338,8 @@ TEST(UdpDriverShould, FailTransmitWhenPbufTakeUnsuccessful) {
 }
 
 TEST(UdpDriverShould, FailTransmitWhenUdpConnectUnsuccessful) {
-    MockUdpInterface udp_if;
-    MockOsInterface os_if;
+
+
     uint8_t txBuff[10] = {};
     struct pbuf txPbuf;
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
@@ -357,8 +356,8 @@ TEST(UdpDriverShould, FailTransmitWhenUdpConnectUnsuccessful) {
 }
 
 TEST(UdpDriverShould, FailTransmitWhenUdpSendUnsuccessful) {
-    MockUdpInterface udp_if;
-    MockOsInterface os_if;
+
+
     uint8_t txBuff[10] = {};
     struct pbuf txPbuf;
     const ip_addr_t TEST_IP_ADDR = {0xC0A80008};
```
</details>

rfairley, 2019-02-07:
Going to close this as we have reproducible results for these errors - will be resolved in #174

closed 2019-02-07 by rfairley